r-lidar/rlas #4: Shift of Z values in writeLAS output

floriandeboissieu, 2017-05-23:
Hi Jean-Romain,
I noticed a 0.01 shift of certain Z values when writing to las/laz, there seem to be a bug there. It seems to be related with some particular Z values, see code below.
Here is how I could reproduce the bug:

> LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
> lidar = readLAS(LASfile)
> writeLAS(lidar,"~/Megaplot.cpy.laz")
365868 bytes written
Column pulseID skipped. It does not meet las specifications.
> lidar1=readLAS("~/Megaplot.cpy.laz")
> any(lidar$Z-lidar1$Z>0)
[1] TRUE
> (lidar$Z-lidar1$Z)[lidar$Z-lidar1$Z>0][1:10]
 [1] 0.01 0.01 0.01 0.01 0.01 0.01 0.01 0.01 0.01 0.01

> writeLAS(lidar1,"~/Megaplot.cpy.cpy.laz")
365868 bytes written
Column pulseID skipped. It does not meet las specifications.
> lidar2=readLAS("~/Megaplot.cpy.cpy.laz")
> any(lidar2$Z-lidar1$Z>0)
[1] FALSE

floriandeboissieu, 2017-05-23:
> LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
> lidar = readLAS(LASfile)
> 
> writeLAS(lidar,"~/Megaplot.cpy.laz")
365868 bytes written
Column pulseID skipped. It does not meet las specifications.
> lidar1=readLAS("~/Megaplot.cpy.laz")
> any(lidar$Z-lidar1$Z>0)
[1] TRUE
> (lidar$Z-lidar1$Z)[lidar$Z-lidar1$Z>0][1:10]
 [1] 0.01 0.01 0.01 0.01 0.01 0.01 0.01 0.01 0.01 0.01
> writeLAS(lidar1,"~/Megaplot.cpy.cpy.laz")
365868 bytes written
Column pulseID skipped. It does not meet las specifications.
> lidar2=readLAS("~/Megaplot.cpy.cpy.laz")
> any(lidar2$Z-lidar1$Z>0)
[1] FALSE

floriandeboissieu, 2017-05-23:
Easier to copy paste like this maybe
```
LASfile <- system.file("extdata", "Megaplot.laz", package="lidR")
lidar = readLAS(LASfile)
writeLAS(lidar,"~/Megaplot.cpy.laz")
lidar1=readLAS("~/Megaplot.cpy.laz")
any(lidar$Z-lidar1$Z>0)
(lidar$Z-lidar1$Z)[lidar$Z-lidar1$Z>0][1:10]
writeLAS(lidar1,"~/Megaplot.cpy.cpy.laz")
lidar2=readLAS("~/Megaplot.cpy.cpy.laz")
any(lidar2$Z-lidar1$Z>0)
```

Jean-Romain, 2017-05-23:
Nice

Jean-Romain, 2017-05-23:
Well, I was all wrong. `set_z` does not take into account the offset and the scale factor. This is not the solution the file is all wrong

Jean-Romain, 2017-05-23:
Try that one. It's fine on my computer

commit ab052b44 (2017-05-23): Fix #4 - bug of double precision
commit f570f0f0 (2017-05-23): Fix #4 - bug of double precision
closed 2017-06-05 by Jean-Romain