r-lidar/rlas #5: [NEWBIE] What is the features that include this R package?

jabrena, 2017-06-09:
Hi, I would like to test your R library.

What is the features that the library include?

Cheers

Juan Antonio

Jean-Romain, 2017-06-09:
Hi,

Well, there are only two functions so you can read the entire documentation very quickly :smile: . Basically:
- Read las and laz files
- Write las and laz files
- Filter las and laz files while reading (streaming filter - do not load useless data in memory)

This package drives the I/O of my `lidR` package. It is fully functional but not appropriated for dummies. You should master the official las specifications if you want to use it properly. Indeed this package strictly aims to read and write las and laz files and do not *think for you*. This is especially true for the `writeLAS` function. My `lidR` package, however is designed to *think for the user* and enable to read and write in a seamless way. It depends of your needs.

Internally `rlas` is a wrapper over a slightly modified version of LASlib used in LAStools and created by Martin Isenburg.

Cheers

jabrena, 2017-06-12:
Do you know some good guide to write laz files?

I would like to generate `laz` files to be processed later with libraries like your R library.

Juan Antonio

Jean-Romain, 2017-06-12:
laz files are compressed las files. What you need to understand las files then is works in a seamless way. There is only one guide to understand what is a las file and it is the [official specifications](http://www.asprs.org/wp-content/uploads/2010/12/LAS_1_4_r13.pdf).

closed 2017-06-27 by Jean-Romain