r-lidar/rlas #13: Wrong File Signature Error

vultronn, 2018-02-02:
I have been unsuccessful to import my LAS files b/c I keep getting the errors...
> readLAS("C:\\myfile.las") # this is an example file path that is structured correctly in my script

ERROR: wrong file signature <'This displays the first two rows of my LAS file'>
ERROR: cannot open lasreaderlas with file name "C:\\myfile.las"
Error: LASlib internal error. See message above.

I can read this into python using lasio, but have had no success in R.  Are there any ideas why I am unsuccessful in importing the data?  I am using the readLAS function.

Jean-Romain, 2018-02-02:
Your file is corrupted. The header of the file (the metadata) should contains a signature that contains "LASF" code. This is required by the las specifications. Here considering your error I guess the file signature is not "LASF". Thus your file is not a proper las file that respect the specifications. And considering your error I guess this file is not a las file at all. Indeed what is weir is the fact it displays the first two rows.

I don't know lasio but from what I quickly read from the website.

> This is a Python 2.7 and 3.3+ package to read and write Log ASCII Standard (LAS) files, used for borehole data such as geophysical, geological, or petrophysical logs. It’s compatible with versions 1.2 and 2.0 of the LAS file specification, published by the Canadian Well Logging Society.

I think you are talking about log ASCII standard. Not laser data las files.

vultronn, 2018-02-02:
Hm, that's very interesting.  I didn't know there was a difference.  I'm new to the las formatting world.  Yeah, it does appear that we use LAS format, but they're not real las files with XYZ values.  Thank you very much!

closed 2018-02-02 by Jean-Romain