r-lidar/rlas #15: Issue reading in LAS 1.4 prf 6 file

AronBoettcher, 2018-02-28:
Trying to read in a LAS 1.4 prf 6 and I'm getting an error: 'LAS format not yet supported'
This seems like expected behavior, but.. I really want to read this file.
Is this expected behavior?

Jean-Romain, 2018-02-28:
Hi, this is an expected behavior because I've never encountered such file type. So, I was not able debug the code and I disabled the support waiting for somebody complaining :smile:. You are the one. Send me a piece of file and I will enable the support of such format.

Jean-Romain, 2018-03-09:
Hi, do you still need to read LAS1.4 prf 6 file?

AronBoettcher, 2018-03-10:
Hello, I had to make sure I had permission to share these data. The attached is a LAS1.4 prf 6 file.
[las14_pdrf6_sample.zip](https://github.com/Jean-Romain/rlas/files/1798867/las14_pdrf6_sample.zip)

Jean-Romain, 2018-03-10:
Thank, I will try to enable the feature tomorrow.

Jean-Romain, 2018-03-10:
Done in read more. write more is more work. Update your `rlas` version from github

closed 2018-03-10 by Jean-Romain

AronBoettcher, 2018-03-12:
Great, I'll give it a run. Thanks!

commit df3aad87 (2018-03-10): Enable point format 6 fix #15