r-lidar/rlas #16: Extra bytes with NAs are not written

Jean-Romain, 2018-03-14:
@floriandeboissieu 

See testthat `test-header tools` line 41.

> Index out of bounds: [index='no_data'].

This come from `writeLAS.cpp` line 137. I do not support properly `no_data` because I do not understand what is the expected information.

floriandeboissieu, 2018-03-14:
no_data is the value that should replace NA value when writing file. 
Example, let say that you have an Extra Byte with integer values that can never reach value -999, well you can set no_data=-999. Before sending the data to laswriter, all NA should be replaced by -999 and when reading the written file, all values at -999 should be replaced by NA before sending it back to user.

Another example: let say that you have an Extra Byte with float values, that you want store in integer after scaling with scale=0.01 and offset=50.2. Let's say 0.0 is a value never reached by the Extra Byte, i.e. no_data=0.0. All NA values should be replaced by 50.2 before calling laswriter,  the no_data value will be scaled in the cpp writer and stored in the file as int((0.0 - 50.2)/0.01)=-5020. At reading, the opposit should happen, i.e. -50.2 values should be replaced by NA values before sending them to user.

Does it answer the question?

floriandeboissieu, 2018-03-14:
I just look into the code, no_data should be part of header_add_extrabytes_manual signature. If has_na=1 a no_data value should be set in list in the same function. By default, it could be the maximum of the data type, e.g. .Machine$integer.max for integers, .Machine$double.xmax for doubles.

Jean-Romain, 2018-03-15:
I got it!

commit ae2921aa (2018-03-15): supports NA fix #16
closed 2018-03-19 by Jean-Romain