r-lidar/rlas #17: CRAN compilation failed on Windows i386

Jean-Romain, 2018-04-03:
@floriandeboissieu: The windows builder gives an error for rlas 1.2.0 on Windows for 32 bits architecture. https://win-builder.r-project.org/mQ1z5qA5JIY5/. This error is likely in testthat. I can't figure out what is the error. Do you have any idea? A way to reproduce?

floriandeboissieu, 2018-04-04:
With so little clue in the log... I'll check tomorrow if I can find 
something

Cheers

floriandeboissieu, 2018-04-04:
Anything with rhub check?

Jean-Romain, 2018-04-04:
Yep an rhub does not provide more informations. It failed after 1h30.

Jean-Romain, 2018-04-04:
https://builder.r-hub.io/status/rlas_1.2.0.tar.gz-42dbc6ac7089497d901ef8aa630423c4

Jean-Romain, 2018-04-05:
I sequentially uploaded 3 versions

1. All the code in `testthat` was commented: :heavy_check_mark: 
2. All the code in `testthat` was commented but `test-head_tools.R` : :x: 
3. All the code in `testthat` was commented but `test-readlas.R` : :heavy_check_mark: 

So the problem seems to come from `test-head_tools.R`. I slowily investigate. One clue each time the CRAN compile the package i.e. once each 15 minutes :smile: 

The big deal is that is that `header_tools` is pure R stuff...

Jean-Romain, 2018-04-05:
The issue occurs at several place but one of them is ligne 18 of `test_header_tools.R`

```r
wheader = read.lasheader(write_path)
```

However line 5 of the same file this one works

```r
header = read.lasheader(lazfile)
```

floriandeboissieu, 2018-04-06:
I started the same yesterday, commenting all but:

     - readlas tests -> works

     - writelas tests -> doesn't work

I commented all writelas including back progressively the tests.

floriandeboissieu, 2018-04-06:
I narrowed to line 43 of test-writelas:

header = read.lasheader(lazfile)

I believe it has something to do with vlrsreader function.

Maybe, it could has something to do with I64 type definition, in 
inst/include/mydefs.hpp (line 61-67)

I'm sorry I cannot go on today...

Hopping it helps

Jean-Romain, 2018-04-06:
You're right. I reached the same conclusion.

This works

```r
lazfile <- system.file("extdata", "example.laz", package="rlas")
header = read.lasheader(lazfile)
```

And this don't

```r
lazfile <- system.file("extdata", "extra_byte.laz", package="rlas")
header = read.lasheader(lazfile)
```

Jean-Romain, 2018-04-06:
I'm slowly converging. The bug comes from `vlrsreader` (you were right but I checked it by security). `readheader.cpp` has not been modified a lot thus I'm pretty sure it comes from this commit: fc0b1419

commit 9430b1bb (2018-04-06): Fix #17
closed 2018-04-06 by Jean-Romain