r-lidar/rlas #22: Failed installing rlas using devtools::install_github("Jean-Romain/rlas", dependencies=TRUE)

johnypark, 2018-05-09:
Hello,

Installing rlas is failing and I can't figure out how to fix it. 
Below is error description:
```
`writeLAS.cpp:130:20:` error: ambiguous conversion for C-style cast from 'Proxy' (aka 'generic_proxy<19>') to 'double'
        scale[i] = (double)(as<List>(description["scale"])[0]);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:63:4: note: candidate function
                        operator bool() const { return ::Rcpp::as<bool>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:64:4: note: candidate function
                        operator int() const { return ::Rcpp::as<int>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:58:26: note: candidate function [with U = double]
                        template <typename U> operator U() const {
                                              ^
writeLAS.cpp:137:21: error: ambiguous conversion for C-style cast from 'Proxy' (aka 'generic_proxy<19>') to 'double'
        offset[i] = (double)(as<List>(description["offset"])[0]);
                    ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:63:4: note: candidate function
                        operator bool() const { return ::Rcpp::as<bool>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:64:4: note: candidate function
                        operator int() const { return ::Rcpp::as<int>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:58:26: note: candidate function [with U = double]
                        template <typename U> operator U() const {
                                              ^
writeLAS.cpp:144:23: error: ambiguous conversion for C-style cast from 'Proxy' (aka 'generic_proxy<19>') to 'double'
        scaled_value=((double)(as<List>(description["no_data"])[0]) - offset[i])/scale[i];
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:63:4: note: candidate function
                        operator bool() const { return ::Rcpp::as<bool>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:64:4: note: candidate function
                        operator int() const { return ::Rcpp::as<int>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:58:26: note: candidate function [with U = double]
                        template <typename U> operator U() const {
                                              ^
writeLAS.cpp:184:23: error: ambiguous conversion for C-style cast from 'Proxy' (aka 'generic_proxy<19>') to 'double'
        scaled_value=((double)(as<List>(description["min"])[0]) - offset[i])/scale[i];
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:63:4: note: candidate function
                        operator bool() const { return ::Rcpp::as<bool>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:64:4: note: candidate function
                        operator int() const { return ::Rcpp::as<int>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:58:26: note: candidate function [with U = double]
                        template <typename U> operator U() const {
                                              ^
writeLAS.cpp:224:23: error: ambiguous conversion for C-style cast from 'Proxy' (aka 'generic_proxy<19>') to 'double'
        scaled_value=((double)(as<List>(description["max"])[0]) - offset[i])/scale[i];
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:63:4: note: candidate function
                        operator bool() const { return ::Rcpp::as<bool>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:64:4: note: candidate function
                        operator int() const { return ::Rcpp::as<int>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:58:26: note: candidate function [with U = double]
                        template <typename U> operator U() const {
                                              ^
writeLAS.cpp:381:26: error: conversion from 'Proxy' (aka 'generic_proxy<19>') to 'double' is ambiguous
          scaled_value = as<List>(description["no_data"])[0];
                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:58:26: note: candidate function [with U = double]
                        template <typename U> operator U() const {
                                              ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:63:4: note: candidate function
                        operator bool() const { return ::Rcpp::as<bool>(get()) ; }
                        ^
/Users/JYP/Library/R/3.3/library/Rcpp/include/Rcpp/vector/generic_proxy.h:64:4: note: candidate function
                        operator int() const { return ::Rcpp::as<int>(get()) ; }
                        ^
writeLAS.cpp:381:24: error: assigning to 'double' from incompatible type 'Proxy' (aka 'generic_proxy<19>')
          scaled_value = as<List>(description["no_data"])[0];
                       ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
7 errors generated.
make: *** [writeLAS.o] Error 1
ERROR: compilation failed for package ‘rlas’
* removing ‘/Users/JYP/Library/R/3.3/library/rlas’
* restoring previous ‘/Users/JYP/Library/R/3.3/library/rlas’
Installation failed: Command failed (1)
```

Experiencing error when using readLAS as well: 
```

las = readLAS(files=filename,filter="")
Error: Package rlas v1.1.10 or superior is requiered.
```
Below is session info:
> sessionInfo()
R version 3.3.3 (2017-03-06)
Platform: x86_64-apple-darwin13.4.0 (64-bit)
Running under: OS X Yosemite 10.10.5

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] lidR_1.5.0 rlas_1.1.5

loaded via a namespace (and not attached):
 [1] Rcpp_0.12.16        lattice_0.20-34     digest_0.6.13       withr_2.1.1         grid_3.3.3         
 [6] R6_2.2.2            git2r_0.18.0        httr_1.2.1.9000     curl_2.8.1          data.table_1.10.4-3
[11] raster_2.6-7        sp_1.2-6            devtools_1.13.4     tools_3.3.3         rsconnect_0.4.3    
[16] memoise_1.1.0      

Any suggestions?

Jean-Romain, 2018-05-09:
Which compiler are you using? You session info states OS X so I guess you are using `clang` as default compiler. Right? If you don't know, re-run `devtools::install_github("Jean-Romain/rlas") and copy paste the compilation directive (all the output).

`rlas` is expected to compile will all major compilers your error is weird.

johnypark, 2018-05-09:
Hi Jean-Romain,

I'm not super familiar with which compiler my OS uses, but I think you are right, it seems to be clang based on the output. Let me paste what I got from re-installing rlas. 

```
> devtools::install_github("Jean-Romain/rlas")
Downloading GitHub repo Jean-Romain/rlas@master
from URL https://api.github.com/repos/Jean-Romain/rlas/zipball/master
Installing rlas
'/Library/Frameworks/R.framework/Resources/bin/R' --no-site-file --no-environ --no-save --no-restore  \
  --quiet CMD INSTALL  \
  '/private/var/folders/ws/0788pbz50w9_5k18cb1fx8zm0000gn/T/RtmpsDMQil/devtools7b857c3dd52/Jean-Romain-rlas-407f325'  \
  --library='/Users/JYP/Library/R/3.3/library' --install-tests 

* installing *source* package ‘rlas’ ...
** libs
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreader_txt.cpp -o LASlib/lasreader_txt.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/fopen_compressed.cpp -o LASlib/fopen_compressed.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswriter_las.cpp -o LASlib/laswriter_las.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreaderbuffered.cpp -o LASlib/lasreaderbuffered.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreadermerged.cpp -o LASlib/lasreadermerged.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswriter_wrl.cpp -o LASlib/laswriter_wrl.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreader_shp.cpp -o LASlib/lasreader_shp.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreader_asc.cpp -o LASlib/lasreader_asc.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreaderpipeon.cpp -o LASlib/lasreaderpipeon.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreader_bil.cpp -o LASlib/lasreader_bil.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lastransform.cpp -o LASlib/lastransform.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswaveform13writer.cpp -o LASlib/laswaveform13writer.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswaveform13reader.cpp -o LASlib/laswaveform13reader.o
clang++ -std=c++11 -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswritercompatible.cpp -o LASlib/laswritercompatible.o
[... cut ...]
                        ^
writeLAS.cpp:381:24: error: assigning to 'double' from incompatible type 'Proxy' (aka 'generic_proxy<19>')
          scaled_value = as<List>(description["no_data"])[0];
                       ^ ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
7 errors generated.
make: *** [writeLAS.o] Error 1
ERROR: compilation failed for package ‘rlas’
* removing ‘/Users/JYP/Library/R/3.3/library/rlas’
* restoring previous ‘/Users/JYP/Library/R/3.3/library/rlas’
Installation failed: Command failed (1)
```

johnypark, 2018-05-09:
I don't know if this helps, but it seems like the compilation works well for other packages. For example,
I tried re-installing Rcpp:
```
install.packages("Rcpp")
Installing package into ‘/Users/JYP/Library/R/3.3/library’
(as ‘lib’ is unspecified)

  There is a binary version available but the source version is later:
      binary  source needs_compilation
Rcpp 0.12.14 0.12.16              TRUE

Do you want to install from sources the package which needs compilation?
y/n: y
installing the source package ‘Rcpp’

trying URL 'https://cran.rstudio.com/src/contrib/Rcpp_0.12.16.tar.gz'
Content type 'application/x-gzip' length 3763400 bytes (3.6 MB)
==================================================
downloaded 3.6 MB

* installing *source* package ‘Rcpp’ ...
** package ‘Rcpp’ successfully unpacked and MD5 sums checked
** libs
clang++ -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include    -fPIC  -Wall -mtune=core2 -g -O2  -c Date.cpp -o Date.o
clang++ -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include    -fPIC  -Wall -mtune=core2 -g -O2  -c Module.cpp -o Module.o
clang++ -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include    -fPIC  -Wall -mtune=core2 -g -O2  -c Rcpp_init.cpp -o Rcpp_init.o
clang++ -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include    -fPIC  -Wall -mtune=core2 -g -O2  -c api.cpp -o api.o
clang++ -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include    -fPIC  -Wall -mtune=core2 -g -O2  -c attributes.cpp -o attributes.o
clang++ -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include    -fPIC  -Wall -mtune=core2 -g -O2  -c barrier.cpp -o barrier.o
clang++ -dynamiclib -Wl,-headerpad_max_install_names -undefined dynamic_lookup -single_module -multiply_defined suppress -L/Library/Frameworks/R.framework/Resources/lib -L/usr/local/lib -o Rcpp.so Date.o Module.o Rcpp_init.o api.o attributes.o barrier.o -F/Library/Frameworks/R.framework/.. -framework R -Wl,-framework -Wl,CoreFoundation
installing to /Users/JYP/Library/R/3.3/library/Rcpp/libs
** R
** inst
** preparing package for lazy loading
** help
*** installing help indices
** building package indices
** installing vignettes
** testing if installed package can be loaded
* DONE (Rcpp)

The downloaded source packages are in
	‘/private/var/folders/ws/0788pbz50w9_5k18cb1fx8zm0000gn/T/RtmpJMUoj4/downloaded_packages’
```

Jean-Romain, 2018-05-09:
You are using an old version of `rlas`. First update `rlas` using the current CRAN version which is `1.2.1`. This should solve all your issues. Also the current `lidR` version is `1.4.2`.

If you really want the very last version of `rlas` (1.2.2) available on github. You must compile it yourself. All the compilation flavors supported by the CRAN work but your compilation way is not part of the CRAN flavors:

```
clang++ -std=c++11
```

Surprisingly your compilation went further than on my machine. You need to force the compilation like that

```
 clang++ -stdlib=libc++
```

To do so you can create a `.R/Makevars` (linux and mac) or `.R/Makevars.win` (windows) file in your home directory that contains:

```
CXX = clang++ -stdlib=libc++
```

johnypark, 2018-05-11:
Hi Jean-Romain, 

I installed the old version because I couldn't compile the latest version on my machine. Anyhow, I followed your advice forcing compilation 
```
 clang++ -stdlib=libc++
```
which worked for getting the lastest lidR, but not the rlas, unfortunately. 
```
> install.packages("rlas")
Installing package into ‘/Users/JYP/Library/R/3.3/library’
(as ‘lib’ is unspecified)

  There is a binary version available but the source version is later:
     binary source needs_compilation
rlas  1.1.5  1.2.1              TRUE

Do you want to install from sources the package which needs compilation?
y/n: y
installing the source package ‘rlas’

trying URL 'https://cran.rstudio.com/src/contrib/rlas_1.2.1.tar.gz'
Content type 'application/x-gzip' length 286099 bytes (279 KB)
==================================================
downloaded 279 KB

* installing *source* package ‘rlas’ ...
** package ‘rlas’ successfully unpacked and MD5 sums checked
** libs
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreader_txt.cpp -o LASlib/lasreader_txt.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/fopen_compressed.cpp -o LASlib/fopen_compressed.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswriter_las.cpp -o LASlib/laswriter_las.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreaderbuffered.cpp -o LASlib/lasreaderbuffered.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreadermerged.cpp -o LASlib/lasreadermerged.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswriter_wrl.cpp -o LASlib/laswriter_wrl.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreader_shp.cpp -o LASlib/lasreader_shp.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreader_asc.cpp -o LASlib/lasreader_asc.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreaderpipeon.cpp -o LASlib/lasreaderpipeon.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lasreader_bil.cpp -o LASlib/lasreader_bil.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/lastransform.cpp -o LASlib/lastransform.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswaveform13writer.cpp -o LASlib/laswaveform13writer.o
clang++ -stdlib=libc++  -I/Library/Frameworks/R.framework/Resources/include -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I/usr/local/include -I/usr/local/include/freetype2 -I/opt/X11/include -I"/Users/JYP/Library/R/3.3/library/Rcpp/include"   -fPIC  -Wall -mtune=core2 -g -O2 -c LASlib/laswaveform13reader.cpp -o LASlib/laswaveform13reader.o
[... cut ...]
Platform: x86_64-apple-darwin13.4.0 (64-bit)
Running under: OS X Yosemite 10.10.5

locale:
[1] en_US.UTF-8/en_US.UTF-8/en_US.UTF-8/C/en_US.UTF-8/en_US.UTF-8

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] rlas_1.1.5 lidR_1.5.0

loaded via a namespace (and not attached):
[1] rsconnect_0.4.3     tools_3.3.3         sp_1.2-6            Rcpp_0.12.16        raster_2.6-7       
[6] grid_3.3.3          data.table_1.10.4-3 lattice_0.20-34    
```

Jean-Romain, 2018-05-11:
Why do you really want `rlas v1.2.2`? `rlas 1.2.1` available on CRAN works. The version `1.2.2` comes with few bug corrections. I will release the version `1.2.2` in few days but until this moment I think that the version `1.2.1` can cover your needs.

To be honest I don't know how to solve your problem. I cannot reproduce this issue. If I were able to reproduce the issue I could fix it, but the compilation with clang works on my machine. I don't have access to a machine running on Mac OS. The only one I have is the [r-hub](https://builder.r-hub.io/) server, and the compilation succeeds for MacOS.

johnypark, 2018-05-13:
I just need a working version of ```rlas```. Getting the latest CRAN version also required compilation, and it was causing the same issue on my machine. 

Anyhow, thanks for checking compiling works on your machine. It may be R version issue (since my current R version is 3.3.3). I tried compiling ```rlas``` and ```lidR``` on another machine (ubuntu, R version 3.4), and it worked. Updating R version on my Mac requires OS upgrade, it may take a while for me. Let me get back after trying compilation on R version 3.5.

Jean-Romain, 2018-05-13:
Try to download Mac OS binaries https://cran.r-project.org/web/packages/rlas/index.html (r-oldrel)

closed 2018-05-17 by Jean-Romain