r-lidar/rlas #24: readLAS: File not supported

FalcoNYCDEP, 2018-07-25:
Hello, 

This is a follow up to my previous issue, incorrectly posted to the lidR issue page (issue #157).  The .las file was originally v1.1 and I attempted to update to v1.4 using rapidlasso's las2las tool.  

Following @bi0m3trics advice, I reinstalled / updated rlas and lidR to the most current versions.  The .las file is still coming back as not supported.  

Here are the specs for my v1.4 las file. 

lasinfo (180620) report for 'C:\Users\gfalco\Documents\R\lidR\las_updated\u_4770066125_2009.las'
reporting all LAS header entries:
  file signature:             'LASF'
  file source ID:             0
  global_encoding:            0
  project ID GUID data 1-4:   00000000-0000-0000-0000-000000000000
  version major.minor:        1.4
  system identifier:          'LAStools (c) by rapidlasso GmbH'
  generating software:        'las2las (version 180620)'
  file creation day/year:     0/0
  header size:                375
  offset to point data:       377
  number var. length records: 0
  point data format:          1
  point data record length:   28
  number of point records:    943948
  number of points by return: 633163 256081 49752 4952 0
  scale factor x y z:         0.01 0.01 0.01
  offset x y z:               0 0 0
  min x y z:                  477000.01 4661250.00 341.21
  max x y z:                  477750.00 4661999.99 397.96
  start of waveform data packet record: 0
  start of first extended variable length record: 0
  number of extended_variable length records: 0
  extended number of point records: 943948
  extended number of points by return: 633163 256081 49752 4952 0 0 0 0 0 0 0 0 0 0 0
the header is followed by 2 user-defined bytes
reporting minimum and maximum for all LAS point record entries ...
  X            47700001   47775000
  Y           466125000  466199999
  Z               34121      39796
  intensity           0        255
  return_number       1          4
  number_of_returns   1          4
  edge_of_flight_line 0          1
  scan_direction_flag 0          1
  classification      1          7
  scan_angle_rank   -20         20
  user_data           0          0
  point_source_ID   422        577
  gps_time 317029.380465 419981.856851
number of first returns:        633163
number of intermediate returns: 60037
number of last returns:         624572
number of single returns:       373824
overview over extended number of returns of given pulse: 373824 405498 144536 20090 0 0 0 0 0 0 0 0 0 0 0
histogram of classification of points:
          717697  unclassified (1)
          223173  ground (2)
            3078  noise (7)

Would you please take a look at my attached las file? 
 
[las_updated.zip](https://github.com/Jean-Romain/rlas/files/2228571/las_updated.zip)

Thank you!

Jean-Romain, 2018-07-25:
As I said file 1.4 are not supported. I'll try to enable the supports as soon as possible (which does not necessarily means soon :disappointed: )

FalcoNYCDEP, 2018-07-25:
I'm sorry Jean, I guess I misunderstood.  So is there a different file type I should attempt to convert to? As I mentioned, the original .las files were v1.1.

Jean-Romain, 2018-07-25:
So far files 1.0 to 1.3 are expected to be supported

Jean-Romain, 2018-07-25:
The file you attached in your first post is perfectly read on my computer (so sorry I was wrong las 1.4 are supported).

What is your problem ?

FalcoNYCDEP, 2018-07-25:
At this point, I am not sure.  When I attempt to read in a .las file (v1.1 or v1.4) I am prompted that there is an error with the file not being supported. 

> las = readLAS("C:/Users/gfalco/Documents/R/lidR/las/u_4770066125_2009.las")
Error: File(s) not supported

Jean-Romain, 2018-07-25:
I think your filename is malformed. The magic Windows paths. I think that the file was not found. The error is the same than for a non supported file. Indeed ofr the reader a non-file or a non-supported file is the same. I should improve that.

What if?  Or any other invalid path only supported by Windows.

```r
las = readLAS("C:\Users\gfalco\Documents\R\lidR\las\u_4770066125_2009.las")
```

FalcoNYCDEP, 2018-07-25:
Thank you, problem solved. It was an issue with file paths as I am operating R on a server so I can handle these larger datasets which is not the environment I am not use to running it under.  

THANK YOU!

Solution:

> las = readLAS("//tsclient/C/Users/gfalco/Documents/R/lidR/las/u_4770066125_2009.las")
> las
class        : LAS (LASF v1.1)
memory       : 72 Mb 
extent       : 477000 , 477750 , 4661250 , 4662000 (xmin, xmax, ymin, ymax)
area         : 562165.7 m² (convex hull)
points       : 943948 points,  NA pulses
density      : 1.68 points/m²,  NA pulses/m²
field names  : X Y Z gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngle UserData PointSourceID 
coord. ref.  : NA

closed 2018-07-25 by FalcoNYCDEP