r-lidar/rlas #25: writeLAS writes a deeply corrupted file

jstrunk001, 2018-12-05:
Hello, I have a specific set of laz files that result in strange behavior when read / write with lidR

```r
library(lidR)
las1 = readLAS("SubArea_012_011.8_6.las")
las1
f = tempfile(fileext = ".las")
writeLAS(las1, f)
las2 = readLAS(f)
las2
```

[SubArea_012_011.8_6.las.zip](https://github.com/Jean-Romain/lidR/files/2647415/SubArea_012_011.8_6.las.zip)

Jean-Romain, 2018-12-05:
Well, I found the issue. But in the meantime I don't know how to fix it and I'm not even sure if I should fix it. Basically the header states to add an offset in the binary file but this offset is not respected. I'm not going through details. The following fixes your issues waiting for a better internal fix.

```r
library(lidR)

las = readLAS("SubArea_012_011.8_6.las")
las@header@PHB$`Header Size` = 235

f = tempfile(fileext = ".las")
writeLAS(las, f)
```

I'm wondering if @floriandeboissieu have a better understanding of the header size management in LASlib

jstrunk001, 2018-12-05:
Ok, thank you JR!

[Forest Service Shield]

Jacob Strunk, PhD

Forest Service - Statistician

Pacific Northwest Research Station

Resource Monitoring & Assessment/FIA Program

Vegetation Monitoring & Remote Sensing Team

P:360-218-4285
jstrunk@fs.fed.us<mailto:jstrunk@fs.fed.us>

3625 93rd Ave SW

Olympia, WA 98512

www.fs.fed.us<http://www.fs.fed.us/>
[USDA Logo]<http://usda.gov/>[Forest Service Twitter]<https://twitter.com/forestservice>[USDA Facebook]<https://www.facebook.com/pages/US-Forest-Service/1431984283714112>

Caring for the land and serving people

This electronic message contains information generated by the USDA solely for the intended recipients. Any unauthorized interception of this message or the use or disclosure of the information it contains may violate the law and subject the violator to civil or criminal penalties. If you believe you have received this message in error, please notify the sender and delete the email immediately.

floriandeboissieu, 2018-12-10:
Sorry for the late answer, I don't have much time this end of year.
I'll dig into it as soon as I can.