r-lidar/rlas #26: readLAS "keep_class" (when greater than 31)

Buckmaster7, 2018-12-31:
First off thank you for this very helpful package.  I'm not sure this is a bug or a known issue but...   

I frequently use LAS files with classification numbers greater than 31.  When I use `readLAS` with a `-keep_class` filter greater than 31 I get a massive failure (R studio closes and must restart).  To clarify this is an example of the code that causes the massive failure:

```r
readLAS("File_123.las", filter = "-keep_class 186")  
```

This same line of code works fine if the "-keep_class" number is <=31.  It would be great if this very helpful code would work for higher classification numbers.

Jean-Romain, 2018-12-31:
After reading the source code the good answer is: *rlas does not catch this error*.

Bug confirmed. Thank you for reporting.

Buckmaster7, 2018-12-31:
> **Caution – This email is from an external source. If you are concerned about this message, please forward it to spam@atco.com<mailto:spam@atco.com> for analysis.**

Thanks Jean-Romain for getting back so quickly!  I’m new to R and new to reporting on message boards so I hope my message made sense.  Happy New Year

Glenn Buckmaster  MSc., P. For
Senior Coordinator
Electricity Transmission & Distribution
Electricity Global Business Unit

C.  780 722 7088  F.  780 420 3650

The information transmitted is intended only for the addressee and may contain confidential, proprietary and/or privileged material.  Any unauthorized review, distribution or other use of or the taking of any action in reliance upon this information is prohibited.  If you receive this in error, please contact the sender and delete or destroy this message and any copies.

Jean-Romain, 2018-12-31:
Bug fixed. Now fail with an error:

```r
library(rlas)
lazfile <- system.file("extdata", "example.laz", package="rlas")
lasdata <- read.las(lazfile, filter = "-keep_class 333")
#> ERROR: cannot keep classification 333 because it is larger than 31
#> Error in C_reader(ifiles, ofile, select, filter, filter_wkt) : 
#>   Filter error see message above. 
```

closed 2018-12-31 by Jean-Romain

Jean-Romain, 2019-01-09:
Are you using LAS 1.4 file format with point format 6? In that case class 186 exist. `rlas` now support LAS 1.4. See #27 . But anyway `filter = "-keep_class 186"` is not supported.

Buckmaster7, 2019-01-09:
Yes – you are correct, that appears to be directly related as to why I ran into problems.  This is a bit of a learning process for me although I’ve worked several times with LAS point cloud files I never clued into the limits on the classification codes.  Although R is a bit of a steep learning curve for me (I’m a SAS guy), I thoroughly enjoy using the tools you have developed.  They are lightning fast - especially compared to working with the point cloud in ARCGIS.  Thanks again!

Glenn Buckmaster  MSc., P. For
Senior Coordinator
Electricity Transmission & Distribution
Electricity Global Business Unit

C.  780 722 7088  F.  780 420 3650

Jean-Romain, 2019-01-09:
> They are lightning fast - especially compared to working with the point cloud in ARCGIS

That makes me happy. I'm working hard on speed improvement. I'm glad to know that I beat ARCGIS :wink:

Jean-Romain, 2019-01-10:
If you are working with LAS 1.4 and format > 6, digging int the source code of LAS lib I found `-keep_extended_classification` and `-drop_extended_classification`

```r
rlas::read.las("las14_prf6.las", filter = "-keep_extended_classification 186")
```

commit d645ba72 (2018-12-31): Fix #26

Former-commit-id: a9c12114d6cd10f21d6484a32609491676a2f72e