r-lidar/rlas #27: LAS 1.4

jfprieur, 2019-01-01:
Hi Jean-Romain,

There is not much to be done (I think it is a liblas problem) but I thought you should be aware that i ran into some strange errors when I use LAS 1.4 formatted files. I tried doing the tiling process with this new data and one tile would be written in the directory, then it would be erased, then the new tile, then erased, etc.

When I converted the LAS to 1.2 format using las2las everything worked fine as usual. This is on lidR v2.

As  I said, I have always had problems with 1.4 format files, I know in Python there is a library called laspy that addresses liblas shortcomings, do not know if there is such a package in R that we could try.

Just thought you should be aware(!) if you were not already in case you get questions. Thanks again for the great R package!

Jean-Romain, 2019-01-05:
`rlas 1.3.0` has a better support of LAS 1.4.  I'm not done yet but at least it should fix your issue.

```r
devtools::install_github("Jean-Romain/rlas", ref = "devel")
```

To support LAS 1.4 as well as LAS <= 1.3 in a consistent way I introduced a minor incompatibility with `lidR <= 2.0.0`. So you must also install `lidR 2.0.1`

```r
devtools::install_github("Jean-Romain/lidR")
```

Also if `lidR 2.0.1` is not buggy with LAS 1.4 the full support of LAS 1.4 will wait for `lidR 2.1.0`. For example `rlas 1.3.0` can read the CRS of a LAS 1.4 file as a WKT string but `lidR` is not able to take advantage of such information to build a 'R compliant' CRS.

Jean-Romain, 2019-01-09:
`rlas 1.3.0` has an almost full support of `LAS 1.4` and point formats 0,1,2,3,6,7,8

Read

- [x] Read the multisprectal information `ScannerChannel` of point format > 6
- [x] Read the `OverlapFlag` bit of point format > 6
- [x] Read the extended `ReturnNumber` and `NumberOfReturns` of point format > 6
- [x] Read the extended `Classification` of point format > 6
- [x] Read the extended `ScanAngle` of point format > 6
- [x] Read the WKT OGC CS information

Write

- [x] Write the multisprectal information `ScannerChannel` of point format > 6
- [x] Write the `OverlapFlag` bit of point format > 6
- [x] Write the extended `ReturnNumber` and `NumberOfReturns` of point format > 6
- [x] Write the extended `Classification`
- [x] Write the extended `ScanAngle` of point format > 6
- [ ] Write the WKT OGC CS (missing write of WKT OGC MATH TRANSFORM)

closed 2019-02-05 by Jean-Romain