r-lidar/rlas #31: Full-waveform support: point types 4, 9 , 10

ericrehm, 2019-01-29:
Our LiDAR is full-waveform.  We would like to read/write, i.e., point type 4 (and 9, 10).

Jean-Romain, 2019-01-29:
Point format 4 9 and 10 are not supported because I've never encounter such file. I hard coded an exit for these formats but in theory these formats should be supported simply by removing the conditional statement. By "supported" I mean just the points not the waveform.

To do so I need a file (tiny) in format 4, 9 or 10. Could you share that? With a FWF file example I could also enable support of the waveform in read mode.

ericrehm, 2019-01-30:
I have sent you a very small Point format 4 LAS file (pulse, 3 returns) and the associated WDP file via email.  (I'm at ULaval too.)  It's the output of a new example that's part of LAStools, specifically, LAStools/LASlib/example/lasexample_write_full_waveform.cpp.   More here:  https://groups.google.com/forum/#!topic/lastools/OZMpTBe7roQ .

Jean-Romain, 2019-01-30:
I pushed add8607 in the `devel` branch with a support of full waveform format 4,5,9,10 (point cloud only). The support of the waveform will take me much more time. Also there is no support for writing format 4,5,9,10. If you are also a user of the lidR package, you must update also this package because this new version of `rlas` brings a support of LAS 1.4 with few minor incompatibilities fixed in  lidR 2.0.1.

Jean-Romain, 2019-01-31:
I pushed a first tentative of FWF support in read mode. It works and is compatible with lidR. There is still some work to achieve but lot of things are still unclear to me in the LAS specification. But anyway it is a great achievement for a single day!

```r
las = rlas::read.las("/home/jr/Téléchargements/out.las", select = "xyzW")
las
#>           X       Y      Z    FullWaveForm
#> 1: 235374.7 5800928 276.50 0,0,0,0,0,0,...
#> 2: 235376.0 5800928 272.84 0,0,0,0,0,0,...
#> 3: 235377.4 5800928 268.72 0,0,0,0,0,0,...

plot(las$FullWaveForm[[1]], type = "l")
```

[image]

Jean-Romain, 2019-02-01:
Great improvements today. I read the FullWaveform + the WaveDataPacket and I made a function for FWF interpretation as the content is not trivial

```r
u = read.las("~/Téléchargements/out.las", select = "W")
h = read.lasheader("~/Téléchargements/out.las")

u

fwf = rlas::fwf_interpreter(h,u)[[1]]
xoffset = min(fwf$X)
yoffset = min(fwf$Y)

color.gradient <- function(x, colors=c("#ffebeb","red"), colsteps=100) 
{
  return(colorRampPalette(colors) (colsteps) [ findInterval(x, seq(min(x),max(x), length.out=colsteps)) ] )
}

col = color.gradient(fwf$Amplitude)

rgl::plot3d(fwf$X-xoffset, fwf$Y-yoffset, fwf$Z, xlim = c(-10,10), ylim = c(-10,10), zlim = c(265,280), col = col)
```

[image]

Jean-Romain, 2019-02-01:
@ericrehm could you share a true FWF data? I'm not able to go further with a single synthetic pulse.

Jean-Romain, 2019-02-01:
[image]

ericrehm, 2019-02-02:
RE: animation

Génial!  /eri

/eric rehm

Professionnel de recherche / Research Associate

CERC sur la télédétection de la nouvelle frontière arctique du Canada

Unité Mixte Internationale Takuvik

http://www.takuvik.ulaval.ca/

418-656-2131 ext. 7677

418-254-7234 (mobile)

Jean-Romain, 2021-04-21:
2 years later, I started to work back on it. I updated everything and fixed some stuff. I will try to provide a support of FWF soon

closed 2021-04-23 by Jean-Romain