r-lidar/rlas #35: read.las – ERROR: 'end-of-file …

jaanusremm, 2019-04-11:
Hi,
Get error if trying to read data from LAZ file. Only a minor part of the data set is successfully read.
‘rlas’ version 1.3.1

```
library(rlas)
las.data <- read.las(files='581536_2018_mets.laz')

#> ERROR: 'end-of-file during chunk with index 0' after 39690 of 1004249 points
```

[581536_2018_mets.laz.zip](https://github.com/Jean-Romain/rlas/files/3069567/581536_2018_mets.laz.zip)

Jean-Romain, 2019-04-11:
Thank you for reporting. On my side I have the same with `rlas 1.3.2`. It looks like a corrupted file but `lastools` is able to read it and parse into `.txt` and to uncompress it to `.las`.

I can investigate but honestly I have few hope to find the issue. It all comes from `LASlib` the C++ library that is internally used to read the files. The only thing I can test is to update `LASlib` but it is a long and boring job.

For now I suggested you to uncompressed your file to `.las` with a recent version of `laszip`. This is what I just did and I was able to correclty read the `.las` file.

jaanusremm, 2019-04-12:
It works. At least on the first try. Thanks :)

Jean-Romain, 2019-04-12:
Should work on the second try as well :wink:. I'll try to update LASlib in May to see if that solves the issue.

closed 2019-04-17 by Jean-Romain

Jean-Romain, 2019-04-17:
Fixed by updating LASlib as I guessed.

jaanusremm, 2019-04-18:
(y)

Jean-Romain, 2019-04-18:
Well it is not currently working on Windows. See #37

commit 46a7d857 (2019-04-17): Update LASlib (fix #35 fix #36)