r-lidar/rlas #37: devtools install_github Windows error

JakubCha, 2019-04-18:
Hello,

I have errors installing rlas using devtools on Windows. Seems like these errors are related in many ways to new version of laslib that @Jean-Romain updated a few hours ago.`
Here's the log:

> -  building 'rlas_1.3.2.tar.gz'
>    
> * installing *source* package 'rlas' ...
> ** libs
> c:/Rtools/mingw_64/bin/g++  -std=gnu++11 -I"E:/R-3.5.3/include" -DNDEBUG -DNDEBUG -DUNORDERED -I. -I../inst/include/ -I"E:/R-3.5.3/library/Rcpp/include" -I"E:/R-3.5.3/library/BH/include"        -O2 -Wall  -mtune=generic -c LASlib/lasreader_txt.cpp -o LASlib/lasreader_txt.o
> In file included from C:/Rtools/mingw_64/x86_64-w64-mingw32/include/combaseapi.h:155:0,
>                  from C:/Rtools/mingw_64/x86_64-w64-mingw32/include/objbase.h:14,
>                  from C:/Rtools/mingw_64/x86_64-w64-mingw32/include/ole2.h:17,
>                  from C:/Rtools/mingw_64/x86_64-w64-mingw32/include/wtypes.h:12,
>                  from C:/Rtools/mingw_64/x86_64-w64-mingw32/include/winscard.h:10,
>                  from C:/Rtools/mingw_64/x86_64-w64-mingw32/include/windows.h:97,
>                  from LASlib/lasreader_txt.cpp:37:
> C:/Rtools/mingw_64/x86_64-w64-mingw32/include/objidlbase.h:864:18: error: macro "Realloc" requires 3 arguments, but only 2 given
>          SIZE_T cb) = 0;
>                   ^
> In file included from ../inst/include/lasdefinitions.hpp:61:0,
>                  from ../inst/include/lasreader.hpp:51,
>                  from ../inst/include/lasreader_txt.hpp:40,
>                  from LASlib/lasreader_txt.cpp:31:
> ../inst/include/laspoint.hpp: In member function 'U64 LASwavepacket::getOffset() const':
> ../inst/include/laspoint.hpp:47:60: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline U64 getOffset() const {return ((U64*)&(data[1]))[0];};
>                                                             ^
> ../inst/include/laspoint.hpp: In member function 'U32 LASwavepacket::getSize() const':
> ../inst/include/laspoint.hpp:48:58: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline U32 getSize() const {return ((U32*)&(data[9]))[0];};
>                                                           ^
> ../inst/include/laspoint.hpp: In member function 'F32 LASwavepacket::getLocation() const':
> ../inst/include/laspoint.hpp:49:63: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline F32 getLocation() const {return ((F32*)&(data[13]))[0];};
>                                                                ^
> ../inst/include/laspoint.hpp: In member function 'F32 LASwavepacket::getXt() const':
> ../inst/include/laspoint.hpp:50:57: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline F32 getXt() const {return ((F32*)&(data[17]))[0];};
>                                                          ^
> ../inst/include/laspoint.hpp: In member function 'F32 LASwavepacket::getYt() const':
> ../inst/include/laspoint.hpp:51:57: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline F32 getYt() const {return ((F32*)&(data[21]))[0];};
>                                                          ^
> ../inst/include/laspoint.hpp: In member function 'F32 LASwavepacket::getZt() const':
> ../inst/include/laspoint.hpp:52:57: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline F32 getZt() const {return ((F32*)&(data[25]))[0];};
>                                                          ^
> ../inst/include/laspoint.hpp: In member function 'void LASwavepacket::setOffset(U64)':
> ../inst/include/laspoint.hpp:54:58: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline void setOffset(U64 offset) {((U64*)&(data[1]))[0] = offset;};
>                                                           ^
> ../inst/include/laspoint.hpp: In member function 'void LASwavepacket::setSize(U32)':
> ../inst/include/laspoint.hpp:55:54: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline void setSize(U32 size) {((U32*)&(data[9]))[0] = size;};
>                                                       ^
> ../inst/include/laspoint.hpp: In member function 'void LASwavepacket::setLocation(F32)':
> ../inst/include/laspoint.hpp:56:64: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline void setLocation(F32 location) { ((F32*)&(data[13]))[0] = location;};
>                                                                 ^
> ../inst/include/laspoint.hpp: In member function 'void LASwavepacket::setXt(F32)':
> ../inst/include/laspoint.hpp:57:51: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline void setXt(F32 xt) {((F32*)&(data[17]))[0] = xt;};
>                                                    ^
> ../inst/include/laspoint.hpp: In member function 'void LASwavepacket::setYt(F32)':
> ../inst/include/laspoint.hpp:58:51: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
>    inline void setYt(F32 yt) {((F32*)&(data[21]))[0] = yt;};
>                                                    ^
> ../inst/include/laspoint.hpp: In member function 'void LASwavepacket::setZt(F32)':
> ../inst/include/laspoint.hpp:59:51: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
[... 63 lines cut ...]
>                  from LASlib/lasreader_txt.cpp:31:
> E:/R-3.5.3/include/R_ext/RS.h:74:37: error: expected identifier before '(' token
>  #define Free(p)        (R_chk_free( (void *)(p) ), (p) = NULL)
>                                      ^
> E:/R-3.5.3/include/R_ext/RS.h:74:47: error: 'parameter' declared as function returning a function
>  #define Free(p)        (R_chk_free( (void *)(p) ), (p) = NULL)
>                                                ^
> E:/R-3.5.3/include/R_ext/RS.h:74:50: error: expected ')' before ',' token
>  #define Free(p)        (R_chk_free( (void *)(p) ), (p) = NULL)
>                                                   ^
> E:/R-3.5.3/include/R_ext/RS.h:74:37: error: expected identifier before '(' token
>  #define Free(p)        (R_chk_free( (void *)(p) ), (p) = NULL)
>                                      ^
> E:/R-3.5.3/include/R_ext/RS.h:74:47: error: 'parameter' declared as function returning a function
>  #define Free(p)        (R_chk_free( (void *)(p) ), (p) = NULL)
>                                                ^
> E:/R-3.5.3/include/R_ext/RS.h:74:50: error: expected ')' before ',' token
>  #define Free(p)        (R_chk_free( (void *)(p) ), (p) = NULL)
>                                                   ^
> make: *** [E:/R-3.5.3/etc/x64/Makeconf:215: LASlib/lasreader_txt.o] Error 1
> ERROR: compilation failed for package 'rlas'
> * removing 'E:/R-3.5.3/library/rlas'
> In R CMD INSTALL
> Error in i.p(...) : 
>   (converted from warning) installation of package ‘C:/.../AppData/Local/Temp/RtmpiOIUjs/file44206689189f/rlas_1.3.2.tar.gz’ had non-zero exit status

Jean-Romain, 2019-04-18:
Yes I know, I sent the package to the CRAN windows builder yesterday and it failed compiling the package on Windows. I did not reproduced myself yet since I don't have Windows. Moreover I don't understand the errors. They look to be related to R stuff not to my code that did not really changed actually. I will investigate.

Jean-Romain, 2019-04-18:
Ok it was not such a big issue. I inverted two lines of code after updating LASlib.  Fixed and tested on CRAN and on rhub. Should be ok for you as well.  0739858

commit 0739858c (2019-04-18): Fix #37 - compilation on windows
closed 2019-04-18 by Jean-Romain