r-lidar/rlas #38: readLAS fail (LASlib internal error)

jfbourdon, 2019-05-02:
I recently updated lidR and I'm now unable to read LAS/LAZ files that worked perfectly before.
I was before on lidR (2.0.0), rlas (1.2.8) and R (3.4.3) and updated to lidR (2.0.2), rlas (1.3.2) and R (3.6.0).

The error that readLAS returns:
`Error: LASlib internal error. See message above.
Error in C_reader(ifiles, ofile, select, filter, filter_wkt) :
  LASlib internal error. See message above.`

jfbourdon, 2019-05-02:
Yes, using rlas directly works, but not lidR... I have tried uninstalling/reinstalling lidR without any success.

Jean-Romain, 2019-05-02:
No problem with lidR. It is not suprising to me since `lidR` is not related to this error. This comes from `rlas`. `readLAS` basically call `rlas::read.las` plus some checks on the content. So what you are describing is very weird. Please show and exact copy of your code and outputs with `reprex`

I tried with `rlas 1.3.2`, `lidR 2.0.2`, `R 3.6.0` on Linux.

``` r
library(lidR)
readLAS("~/Téléchargements/clip_LiDAR.laz")
#> class        : LAS (LASF v1.2)
#> point format : 1
#> memory       : 999.3 Kb 
#> extent       :294923, 294943, 5281064, 5281084 (xmin, xmax, ymin, ymax)
#> coord. ref.  : +init=epsg:2949 +proj=tmerc +lat_0=0 +lon_0=-70.5 +k=0.9999 +x_0=304800 +y_0=0 +ellps=GRS80 +towgs84=0,0,0,0,0,0,0 +units=m +no_defs 
#> area         : 312.1502 m² (convex hull)
#> points       : 12599 points
#> density      : 40.36 points/m²
#> names        : X Y Z gpstime Intensity ReturnNumber NumberOfReturns ScanDirectionFlag EdgeOfFlightline Classification Synthetic_flag Keypoint_flag Withheld_flag ScanAngleRank UserData PointSourceID
```

<sup>Created on 2019-05-02 by the [reprex package](https://reprex.tidyverse.org) (v0.2.1)</sup>

jfbourdon, 2019-05-02:
It now works as it should. The problem was probably caused by a conflict between different versions of R and packages. Also, I was working inside a Rproject created with an other version of R and I was getting other error when trying to install packages. So the problem had probably nothing to do with rlas nor lidR.

jfbourdon, 2019-05-02:
I justed found the real root cause by accident... the filepath and filename must not contain accent or other similar special caracter. In my case, I had a long path with some `à` and `é`. I remove those and the file is correctly read.

Jean-Romain, 2019-05-02:
Well, accents should be supported. Look my path in previous post.

closed 2019-05-02 by Jean-Romain