r-lidar/rlas #46: readLAS crashes when reading ALS las files, after previously working

RandCauthon, 2020-03-02:
I have previously been using the lidR package and it was working great (great work!). However today I simply tried to open an airborne las file and the R sessions crashed. Here is the code, it ain't complicated.

library(lidR)
las = readLAS("C:/Users/josep/OneDrive/Documents/Image_fusioning/Las_testing/Las_files/e1054n0839.las")

I have uploaded the example las file here - https://drive.google.com/open?id=1EUn0jErmrM1a7za0k70pPa_MWuxMO8o1

This crashes the R session, either in R or R studio (grey bomb). I have tried to open multiple airborne LiDAR las files from this same dataset and they all crash R. However, when I open much larger las files (100 times larger) from a different project they load into R just fine. I have updated my R version to 3.6.2, updated to the latest version of lidR, checked my filepaths, had the las file on my C drive, and on an external hard drive, doesn't matter. I have checked the las files in other programs such as PDQ viewer and ArcGIS and they look fine and perform fine in those software programs. And like I said everything was working fine yesterday.  I would also note that I had similar experience on a different computer, when I tried to readLAS or put into a las catalog the session crashed. I would be happy to learn that this behavior is due to something I, the user, is doing.

Jean-Romain, 2020-03-02:
This file is read on my computer GNU/linux, `rlas v1.3.5`.  Please check your version of `rlas`.

RandCauthon, 2020-03-02:
Thanks for the quick reply. Both rlas and lidR are updated versions - 
other attached packages:
[1] lidR_2.2.2    raster_3.0-12 sp_1.4-0      rlas_1.3.5 

I am sorry this is not reproducing.

Jean-Romain, 2020-03-02:
So it is a Windows or Mac specific issue. Please report your `sessionInfo()`

RandCauthon, 2020-03-02:
R version 3.6.2 (2019-12-12)
Platform: x86_64-w64-mingw32/x64 (64-bit)
Running under: Windows 10 x64 (build 17763)

Matrix products: default

Random number generation:
 RNG:     Mersenne-Twister 
 Normal:  Inversion 
 Sample:  Rounding 
 
locale:
[1] LC_COLLATE=English_United States.1252  LC_CTYPE=English_United States.1252    LC_MONETARY=English_United States.1252
[4] LC_NUMERIC=C                           LC_TIME=English_United States.1252    

attached base packages:
[1] stats     graphics  grDevices utils     datasets  methods   base     

other attached packages:
[1] lidR_2.2.2    raster_3.0-12 sp_1.4-0      rlas_1.3.5   

loaded via a namespace (and not attached):
[1] compiler_3.6.2    tools_3.6.2       Rcpp_1.0.3        codetools_0.2-16  grid_3.6.2        data.table_1.12.8
[7] lattice_0.20-38

Jean-Romain, 2020-03-02:
I read your file on Windows 7, R 3.5.3 32 bits rlas 1.3.5, lidR 2.2.2 

```r
x = rlas::read.las("H://issue46/e1054n0839.las") # ok
x = lidR::readLAS("H:/Téléchargements/issue46/e1054n0839.las") # ok
```

Then  on Windows 7, R 3.6.1 64 bits rlas 1.3.5, lidR 2.2.2

```r
x = rlas::read.las("H://issue46/e1054n0839.las") # ok
x = lidR::readLAS("H:/Téléchargements/issue46/e1054n0839.las") # crash
x = rlas:::stream.las("H://Téléchargements/issue46/e1054n0839.las") # ok
```
I confirm the bug. However on Windows tools to debug such kind of trouble do not exist so I don't know how and when I will be able to fix it.

Other tests

```r
x = rlas:::stream.las("H://Téléchargements/issue46/e1054n0839.las") # ok
x = rlas:::C_reader("H:/Téléchargements/issue46/e1054n0839.las", "", select = "*", filter = "", filter_wkt = ) # ok
```

I'm moving the question to `lidR` it seem it comes from it. Minimal reproducible example:

```r
data =  rlas::read.las("H://Téléchargements/issue46/e1054n0839.las")                   
header= rlas::read.lasheader("H:/Téléchargements/issue46/e1054n0839.las")
las = lidR::LAS(data, header) # crash
```