r-lidar/rlas #50: Invalid reading of lax file on windows

Jean-Romain, 2021-02-12:
Based on https://github.com/Jean-Romain/lidR/issues/408 and using the las + lax files reported there. This issue is only reproducible on Windows

Expected behaviour (tested on R 4.0.3, Linux, `rlas` 1.3.9)

```r
las <- rlas::read.las('issue 408/157668.5_243478.3.las', filter = "-inside 157980 243480 158270 243770")
dim(las)
#> [1] 8143683      18
```

Observed behaviour (testes on R 4.0.3, Windows, `rlas` 1.3.9)

```r
las <- rlas::read.las('issue 408/157668.5_243478.3.las', filter = "-inside 157980 243480 158270 243770")
dim(las)
#> [1] 30595    18
```
Oberservation:

- 90% of the region of interest is missing. Look like only one quadrant was loaded
- when deleting the lax files it works. This confirms it is an issue with the spatial index
- creating a new index with `rlas::writelax()` does not help
- creating a new index with `lasindex` does not help

One last thing to test on windows with LAStools (tested on Linux)

```
las2las -i 157668.5_243478.3.las -inside 157980 243480 158270 243770 -o clip.las
lasinfo clip.las
#> extended number of point records: 8143683
```

adrienmichez, 2021-02-12:
Here is the result of `lasinfo`:

Note that not all of LAStools is "free" (see http://lastools.org/LICENSE.txt)
contact 'martin.isenburg@rapidlasso.com' to clarify licensing terms if needed.
lasinfo -i "C:\TMP\lidR_bug\transfer_85074_files_52cf1058\clip.las"
lasinfo (210128) report for 'C:\TMP\lidR_bug\transfer_85074_files_52cf1058\clip.las'
reporting all LAS header entries:
  file signature:             'LASF'
  file source ID:             0
  global_encoding:            17
  project ID GUID data 1-4:   54ED7E4E-227B-4221-0000-000000000000
  version major.minor:        1.4
  system identifier:          'LAStools (c) by rapidlasso GmbH'
  generating software:        'las2las (version 210128)'
  file creation day/year:     64/2020
  header size:                375
  offset to point data:       1502
  number var. length records: 1
  point data format:          6
  point data record length:   30
  number of point records:    0
  number of points by return: 0 0 0 0 0
  scale factor x y z:         0.01 0.01 0.01
  offset x y z:               0 0 0
  min x y z:                  157980.00 243480.00 343.38
  max x y z:                  158269.99 243769.99 562.44
  start of waveform data packet record: 0
  start of first extended variable length record: 0
  number of extended_variable length records: 0
  extended number of point records: 8143683
  extended number of points by return: 5292612 2159802 579044 99791 11526 890 18 0 0 0 0 0 0 0 0
variable length header record 1 of 1:
  reserved             0
  user ID              'LASF_Projection'
  record ID            2112
  length after header  1071
  description          'OGC WKT Coordinate System'
    WKT OGC COORDINATE SYSTEM:
    COMPD_CS["NAD83(2011) / Puerto Rico and Virgin Is. + PRVD02 height - Geoid12B (metre)",PROJCS["NAD83(2011) / Puerto Rico and Virgin Is.",GEOGCS["NAD83(2011)",DATUM["NAD83_National_Spatial_Reference_System_2011",SPHEROID["GRS 1980",6378137,298.257222101,AUTHORITY["EPSG","7019"]],AUTHORITY["EPSG","1116"]],PRIMEM["Greenwich",0,AUTHORITY["EPSG","8901"]],UNIT["degree",0.0174532925199433,AUTHORITY["EPSG","9122"]],AUTHORITY["EPSG","6318"]],PROJECTION["Lambert_Conformal_Conic_2SP"],PARAMETER["standard_parallel_1",18.43333333333333],PARAMETER["standard_parallel_2",18.03333333333333],PARAMETER["latitude_of_origin",17.83333333333333],PARAMETER["central_meridian",-66.43333333333334],PARAMETER["false_easting",200000],PARAMETER["false_northing",200000],UNIT["metre",1,AUTHORITY["EPSG","9001"]],AXIS["X",EAST],AXIS["Y",NORTH],AUTHORITY["EPSG","6566"]],VERT_CS["PRVD02 height - Geoid12B (metre)",VERT_DATUM["Puerto Rico Vertical Datum of 2002",2005,AUTHORITY["EPSG","1123"]],UNIT["metre",1,AUTHORITY["EPSG","9001"]],AXIS["Gravity-related height",UP],AUTHORITY["EPSG","6641"]]]
the header is followed by 2 user-defined bytes
reporting minimum and maximum for all LAS point record entries ...
  X            15798000   15826999
  Y            24348000   24376999
  Z               34338      56244
  intensity           0      65535
  return_number       1          7
  number_of_returns   1          7
  edge_of_flight_line 0          1
  scan_direction_flag 0          1
  classification      1          2
  scan_angle_rank   -29         29
  user_data           0          1
  point_source_ID    10      15054
  gps_time 210857022.314920 222951009.946351
  extended_return_number          1      7
  extended_number_of_returns      1      7
  extended_classification         1      2
  extended_scan_angle         -4781   4832
  extended_scanner_channel        0      1
number of first returns:        5292612
number of intermediate returns: 690938
number of last returns:         5295377
number of single returns:       3135244
overview over extended number of returns of given pulse: 3135244 3160243 1436711 352917 53209 5236 123 0 0 0 0 0 0 0 0
histogram of classification of points:
         7741938  unclassified (1)
          401745  ground (2)
 +-> flagged as extended overlap: 1986
  +--->            1871 of those are unclassified (1)
  +--->             115 of those are ground (2)

Jean-Romain, 2021-02-12:
I tried with `rlas` 1.3.7, 1.3.9 and 1.4.0 either compiling in C++11 and C++14. They all failed to perform the query on Windows. So the bug actually exist for a long time. I can also reproduce with first return only:

```bash
las2las -first_only -i 157668.5_243478.3.las -o test.las
lasindex -i test.las
```

Linux

```r
las <- rlas::read.las('~/Téléchargements/issue 408/test.las', filter = "-inside 157980 243480 158270 243770")
dim(las)
#> [1] 5292612      18
```

Windows

```r
las <- rlas::read.las('~/Téléchargements/issue 408/test.las', filter = "-inside 157980 243480 158270 243770")
dim(las)
#> [1] 5292612      18

adrienmichez, 2021-02-13:
Does that mean that during the mean time, we should not use spatial indexing??

Jean-Romain, 2021-02-14:
Have you ever encountered other issues on windows ? You are the first one to report an problem and so far we were only able to reproduce on this file. It is definitively a complex problem that needs much more investigation.

adrienmichez, 2021-02-15:
Sure... But I don't want to manually check if something went wrong after using spatial indexing. I hope you will soon find how to fix this :-)

Jean-Romain, 2021-02-15:
Bad new for today. I've updated `rlas` with the latest source code of `LASlib` and `LASzip` and it is still failing :disappointed:

Jean-Romain, 2021-02-16:
I made very little progress today. It seems that the coordinates of some points are badly read for unkown reasons. This starts at interval [72200247, 72517435]

### Linux

```
> las <- rlas::read.las('~/Téléchargements/issue 50/157668.5_243478.3.las', filter = "-inside 157980 243480 158270 243770")
16 cells of 156900/242600 158500/244200 intersect rect 157980/243480 158270/243770
LASindex: used 16 cells of total 182
Interval start = 2411200, end = 2678575, full = 15060790, total = 16622934, cells = 16 
   Point (157933.73, 243496.61) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157980.84, 243478.93) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 5084464, end = 5285031, full = 15060790, total = 16622934, cells = 16 
   Point (157929.04, 243443.84) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157958.77, 243464.21) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 7596484, end = 7819087, full = 15060790, total = 16622934, cells = 16 
   Point (157915.35, 243425.31) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157971.41, 243410.94) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 9948801, end = 9968545, full = 15060790, total = 16622934, cells = 16 
Interval start = 9975888, end = 10007926, full = 15060790, total = 16622934, cells = 16 
   Point (157902.24, 243378.48) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 10014607, end = 10045098, full = 15060790, total = 16622934, cells = 16 
Interval start = 10051854, end = 10081465, full = 15060790, total = 16622934, cells = 16 
Interval start = 10089090, end = 10118304, full = 15060790, total = 16622934, cells = 16 
   Point (157970.58, 243399.31) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 10126859, end = 10151486, full = 15060790, total = 16622934, cells = 16 
Interval start = 10161245, end = 10191604, full = 15060790, total = 16622934, cells = 16 
Interval start = 72200247, end = 72517435, full = 15060790, total = 16622934, cells = 16 
   Point (157915.09, 243775.11) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157950.81, 243804.03) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157977.76, 243773.96) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 74745515, end = 75102684, full = 15060790, total = 16622934, cells = 16 
   Point (157890.74, 243766.12) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157946.23, 243765.06) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157963.39, 243742.14) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 77246009, end = 77609879, full = 15060790, total = 16622934, cells = 16 
   Point (157910.84, 243733.35) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157926.59, 243732.23) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157963.89, 243737.67) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 80070354, end = 80392642, full = 15060790, total = 16622934, cells = 16 
   Point (157900.10, 243693.55) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157944.15, 243688.27) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157961.20, 243692.20) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 83094087, end = 83456284, full = 15060790, total = 16622934, cells = 16 
   Point (157902.93, 243668.63) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157931.57, 243666.55) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157969.85, 243664.62) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 86199040, end = 86575969, full = 15060790, total = 16622934, cells = 16 
   Point (157903.58, 243616.46) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157925.78, 243621.48) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157968.21, 243635.94) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 89298031, end = 89650155, full = 15060790, total = 16622934, cells = 16 
   Point (157907.23, 243580.75) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157932.89, 243609.17) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157970.16, 243581.41) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 92151662, end = 92501280, full = 15060790, total = 16622934, cells = 16 
   Point (157893.44, 243554.00) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157939.32, 243559.21) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157974.29, 243567.49) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 95015614, end = 95316926, full = 15060790, total = 16622934, cells = 16 
   Point (157912.87, 243520.20) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (157963.63, 243530.99) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Interval start = 96375793, end = 97411041, full = 15060790, total = 16622934, cells = 16 
   Point (157944.13, 243512.19) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (158043.70, 243473.14) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
[... cut ...]
   Point (1522403.68, -13455192.45) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (2670594.40, -13392933.25) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (2305558.88, -12499677.57) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (2425489.76, -11671302.53) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
   Point (2627340.64, -11566444.93) not in rect [157980.00, 243480.00, 158270.00, 243770.00]
Read 16622934 points
Read 30595 points in bbox
```

adrienmichez, 2021-03-02:
Hi,

I would like to use spatial indexing again but I'm afraid to experience this bug again and not see it.

Is there a way to test if the bug occurred in my process?

Jean-Romain, 2021-03-02:
I'm closer and closer from the bug. I reached very deep lines of code but I'm not yet sure were does it comes from. I cannot say anything about this bug expect that I'm sure the lax files is properly read. What happens is that after a certain number of file cursor moves it seems that there is a lag of few bits between where the cursor should be and where it actually is. It is windows specific and never reproduced with another dataset. Your dataset is very big and this does not help to figure out the issue.

It is not reproducible in all OS, not reproducible in other dataset, not reproducible with LAStools, not alway reproducible with another spatial query. My best guess so far is that it is specific to the c++ compiler used by R but cannot explain why only with this dataset.

But I'm still investigating regularly getting closer and closer to the problem.

Jean-Romain, 2021-03-02:
Good new I finally identified the issue :+1: . Now let try to fix it.

adrienmichez, 2021-03-02:
Wow... Great news!

Jean-Romain, 2021-03-02:
I fixed it :tada:.

That was a big issue actually and I'm surprised nobody reported it before. I'm tagging  @bi0m3trics wondering if you already encountered such problem.

To make it simple the problem came from the fact that your file was more than ~2GB. This is not expected to be a problem to read such big file in the regular case. However when using a lax file, the reader can jump to different parts of the files. If the file is bigger than ~2GB it may try to jump to positions bigger than 2147483647. The problem was that the code badly handled the case when compiled with `mingw` which is the compiler used by R on windows. It assigned a method to read the that cannot handle more than 2GB. The points that were missing in our query were those in the file after 2GB.

adrienmichez, 2021-03-02:
Great 
Thanks for the investigation!

Jean-Romain, 2021-03-02:
Thank you for reporting. Please try `rlas 1.4.0` on your side and give me some feed back. I fixed the issue in fea7e62 but I will also made a full update of `LASlib` in `rlas 1.4.0`. So please stay informed. The current version 1.4.0 will be outdated in few hours/days

adrienmichez, 2021-03-04:
It's now working using `lidR` and `rlas` from github 🎉

commit fea7e626 (2021-03-02): Fix #50 off64_t
closed 2021-03-04 by Jean-Romain