r-lidar/rlas #52: Unable to install rlas dependency on linux

mxblsdl, 2021-04-06:
I have been trying to install `lidR` on RStudio Server and run into a error on the `rlas` dependency installation. The error actually fills the console beyond the character limit so I can't share the entire error or see where it begins. The error repeats similar lines to;
```
 required from here
/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include/boost/geometry/core/tag.hpp:68:17: 
error: no type named ‘type’ in ‘struct 
boost::geometry::util::remove_cptrref<boost::geometry::model::multi_polygon<boost::geometry::model::
polygon<boost::geometry::model::point<double, 2ul, boost::geometry::cs::cartesian> > > >’
```
Over and over. I'm not familiar with the `BH` package but all of the errors seem to stem from an issue there. System info:
```
platform       x86_64-pc-linux-gnu         
arch           x86_64                      
os             linux-gnu                   
system         x86_64, linux-gnu           
status                                     
major          4                           
minor          0.2                         
year           2020                        
month          06                          
day            22                          
svn rev        78730                       
language       R                           
version.string R version 4.0.2 (2020-06-22)
nickname       Taking Off Again  
```
My Makevars file only includes the line:
```
CXX14 = g++ -std=c++1y
```
Wondering if you have ever seen an error like this or have any idea what could be causing it. Any advice would be appreciated.

mxblsdl, 2021-04-06:
When I remove that line from the Makevar file I get the error
```
Error: C++14 standard requested but CXX14 is not defined
```

Jean-Romain, 2021-04-06:
Ok that's an error already seen on Windows for old versions of R. But you said you are using R 4.0. Which linux distribution are you using? Which version of g++ is used by default? Can you show more of the output of the compilation?

mxblsdl, 2021-04-06:
Running `g++ -v` outputs:
```
Using built-in specs.
COLLECT_GCC=g++
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/4.8.5/lto-wrapper
Target: x86_64-redhat-linux
Configured with: ../configure --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-bootstrap --enable-shared --enable-threads=posix --enable-checking=release --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-linker-hash-style=gnu --enable-languages=c,c++,objc,obj-c++,java,fortran,ada,go,lto --enable-plugin --enable-initfini-array --disable-libgcj --with-isl=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/isl-install --with-cloog=/builddir/build/BUILD/gcc-4.8.5-20150702/obj-x86_64-redhat-linux/cloog-install --enable-gnu-indirect-function --with-tune=generic --with-arch_32=x86-64 --build=x86_64-redhat-linux
Thread model: posix
gcc version 4.8.5 20150623 (Red Hat 4.8.5-44) (GCC) 
```

Using red hat distribution. The compilation error is only:
```
* installing *source* package ‘rlas’ ...
** package ‘rlas’ successfully unpacked and MD5 sums checked
** using staged installation
** libs
Error: C++14 standard requested but CXX14 is not defined
* removing ‘/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/rlas’
Warning in install.packages :
  installation of package ‘rlas’ had non-zero exit status
``` 
When I install without a Makevar file.

Jean-Romain, 2021-04-06:
gcc 4.8 is pretty old and does not support C++14. Only pre-standard C++14 with `c++1y`. See https://stackoverflow.com/questions/31965413/compile-c14-code-with-g

The CRAN no longer compiles with gcc 4.8. They used 4.9 for old versions of R and gcc > 8 on all other flavours. See https://cran.r-project.org/web/checks/check_flavors.html#r-devel-linux-x86_64-fedora-gcc

So the way is to get a fairly recent compiler I guess.

mxblsdl, 2021-04-06:
Okay thank you this is very helpful. I don't have direct control over the operating system so updating the compiler will have to go through IT. I will follow up if this provides a solution.

mxblsdl, 2021-04-23:
I got `gcc` updated to version 7.3.1 but I'm still getting errors when trying to compile `rlas`. 

A sample of the errors:
```
/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include/boost/geometry/core/tag.hpp:68:17: error: no type named ‘type’ in ‘struct boost::geometry::util::remove_cptrref<boost::geometry::model::multi_polygon<boost::geometry::model::polygon<boost::geometry::model::point<double, 2ul, boost::geometry::cs::cartesian> > > >’
```

Using the Makevar file:
```
CXX14=gcc
PKG_CXXFLAGS = -std=gnu++1y
```

If I don't use a Markvar file I get the error:
`Error: C++14 standard requested but CXX14 is not defined`

Any ideas on what could be causing this or how to get around it?

Jean-Romain, 2021-04-23:
Remove your `PKG_CXXFLAGS` and check if `gcc` really points to gcc7

```
gcc -v
```

if not check if `gcc-7` is an existing command.

For example on my computer `gcc` points to `gcc-9` which is `9.3.0` but I also have `gcc-10`. Then you have two options. The best is to make `gcc` pointing to `gcc-7` instead of `gcc-4`. Google it I don't know how to do it without google. Or you can

```
CXX14=gcc-7
```

mxblsdl, 2021-04-23:
I changed the `CXX14` variable to point to the newer version of `gcc`. To install `gcc` on RHEL7 we needed to use devtools-set and it puts the newer version of `gcc` in a different subfolder.

Unfortunately I'm getting a different error now:
```
./RcppExports.o: relocation R_X86_64_32 against `.bss' can not be used when making a shared object; recompile with -fPIC

/opt/rh/devtoolset-7/root/usr/libexec/gcc/x86_64-redhat-linux/7/ld: final link failed: Nonrepresentable section on output
```
Any ideas on this?

Jean-Romain, 2021-04-23:
Absolutely not. Googling it gives a lot of results that seems to be related to specific instruction set of processor architecture... if correct this is waaaay too low level for me. But some stackoverflow answers are in accordance with the message you get:

> recompile with -fPIC

In the Makevars try to add 

```
CXX14FLAGS=-fPIC
```

Please copy/paste the compilation directives you see on screen to see everything that happened during compilation

mxblsdl, 2021-04-26:
I moved the `-fPIC` flag to the line: `CXX14 = /opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC`

And I'm almost able to load the package. I now get a different error shown below. There are some leads by googling this error but still haven't figured it out. Seems like an issue with `_ZNKSt9basic_iosIcSt11char_traitsIcEEcvbEv` but I'm not sure what that is.
```
*** installing help indices
  converting help for package ‘rlas’
    finding HTML links ... done
    check                                   html  
    crs_tools                               html  
    extra_bytes_attribute_tools             html  
    las_specification_tools                 html  
    public_header_block_tools               html  
    read.las                                html  
    read.lasheader                          html  
    write.las                               html  
    writelax                                html  
** building package indices
** testing if installed package can be loaded from temporary location
Error: package or namespace load failed for ‘rlas’ in dyn.load(file, DLLpath = DLLpath, ...):
 unable to load shared object '/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/00LOCK-rlas/00new/rlas/libs/rlas.so':
  /home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/00LOCK-rlas/00new/rlas/libs/rlas.so: undefined symbol: _ZNKSt9basic_iosIcSt11char_traitsIcEEcvbEv
Error: loading failed
Execution halted
ERROR: loading failed
```

Jean-Romain, 2021-04-26:
Show me the whole compilation output

mxblsdl, 2021-04-26:
```
Installing package into ‘/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0’
(as ‘lib’ is unspecified)
trying URL 'https://cran.rstudio.com/src/contrib/rlas_1.4.0.tar.gz'
Content type 'application/x-gzip' length 387172 bytes (378 KB)
==================================================
downloaded 378 KB

* installing *source* package ‘rlas’ ...
** package ‘rlas’ successfully unpacked and MD5 sums checked
** using staged installation
** libs
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lasreader_txt.cpp -o LASlib/lasreader_txt.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/fopen_compressed.cpp -o LASlib/fopen_compressed.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/laswriter_las.cpp -o LASlib/laswriter_las.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lasreaderbuffered.cpp -o LASlib/lasreaderbuffered.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lasreadermerged.cpp -o LASlib/lasreadermerged.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/laswriter_wrl.cpp -o LASlib/laswriter_wrl.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lasreader_shp.cpp -o LASlib/lasreader_shp.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lasreader_asc.cpp -o LASlib/lasreader_asc.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lasreaderstored.cpp -o LASlib/lasreaderstored.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lasreaderpipeon.cpp -o LASlib/lasreaderpipeon.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lasreader_bil.cpp -o LASlib/lasreader_bil.o
/opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include      -c LASlib/lastransform.cpp -o LASlib/lastransform.o
LASlib/lastransform.cpp:791:149: warning: multi-character character constant [-Wmultichar]
   inline I32 get_command(CHAR* string) const { return sprintf(string, "-%s%c %u ", name(), (rgbi == 0 ? 'R' : (rgbi == 1 ? 'G' : (rgbi == 2 ? 'B' : 'NIR'))), index); };
                                                                                                                                                     ^~~~~
LASlib/lastransform.cpp:1069:149: warning: multi-character character constant [-Wmultichar]
[... cut ...]
Error: package or namespace load failed for ‘rlas’ in dyn.load(file, DLLpath = DLLpath, ...):
 unable to load shared object '/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/00LOCK-rlas/00new/rlas/libs/rlas.so':
  /home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/00LOCK-rlas/00new/rlas/libs/rlas.so: undefined symbol: _ZNKSt9basic_iosIcSt11char_traitsIcEEcvbEv
Error: loading failed
Execution halted
ERROR: loading failed
* removing ‘/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/rlas’
Warning in install.packages :
  installation of package ‘rlas’ had non-zero exit status
```

Jean-Romain, 2021-04-26:
Could you remember me why you need to use a custom `Makevars`? I don't know why Red Hat does not work like every other linux distros. My compilation directives, which are the default not overwritten by a `Makevars`, are:

```
g++ -std=gnu++14 -I"/usr/share/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/jr/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/jr/R/x86_64-pc-linux-gnu-library/4.0/BH/include'    -fpic  -g -O2 -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.5=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -g  -c
```
So you miss at least 

- `std=gnu++14` I guess you are compiling in C++14 anyway otherwise it would have failed earlier
-  `-O2` very important, otherwise compiled C++ will be very slow.
- ` -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.5=.` ???
-  `-fstack-protector-strong` ???
-  `-Wformat`  `-Werror=format-security` `-Wdate-time` probably not important
-  `-D_FORTIFY_SOURCE=2` 
-  `-g` [debbuger](https://www.rapidtables.com/code/linux/gcc/gcc-g.html). I don't know why, I don't know if it matters

Jean-Romain, 2021-04-26:
The linkage is made with

```g++ -std=gnu++14 -Wl,-S -shared -L/usr/lib/R/lib -Wl,-Bsymbolic-functions -Wl,-z,relro -o rlas.so <file.o> -L/usr/lib/R/lib -lR
```

mxblsdl, 2021-04-26:
If I don't include a Makevar file I get the error:
`Error: C++14 standard requested but CXX14 is not defined`

I tried the `fPIC` in `CXX14FLAGS` and compilation still failed. [This post](https://stackoverflow.com/questions/332767/recompile-with-fpic-option-but-the-option-is-already-in-the-makefile) was helpful for that.

Do you think I should add some of the other compilation variables into Makevars?

Jean-Romain, 2021-04-26:
Yes put the same one than mine

mxblsdl, 2021-04-26:
My Makevar file is now:
```
CXX14 = /opt/rh/devtoolset-7/root/usr/bin/gcc /opt/rh/devtoolset-7/root/usr/bin/g++ -fPIC
CXX14FLAGS = g++ -std=gnu++14 -I"/usr/share/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/R/x86_64-pc-linux-gnu-library/4.0/BH/include'    -fpic  -g -O2 -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -g  -c
```
But I get the error:

```
* installing *source* package ‘rlas’ ...
** package ‘rlas’ successfully unpacked and MD5 sums checked
** using staged installation
** libs
/opt/rh/devtoolset-7/root/usr/bin/gcc /opt/rh/devtoolset-7/root/usr/bin/g++ -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include     -std=gnu++14 -I"/usr/share/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/R/x86_64-pc-linux-gnu-library/4.0/BH/include'    -fpic  -g -O2 -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -g  -c -c LASlib/lasreader_txt.cpp -o LASlib/lasreader_txt.o
gcc: warning: /opt/rh/devtoolset-7/root/usr/bin/g++: linker input file unused because linking not done
/opt/rh/devtoolset-7/root/usr/bin/gcc /opt/rh/devtoolset-7/root/usr/bin/g++ -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include     -std=gnu++14 -I"/usr/share/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/R/x86_64-pc-linux-gnu-library/4.0/BH/include'    -fpic  -g -O2 -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -g  -c -c LASlib/fopen_compressed.cpp -o LASlib/fopen_compressed.o
gcc: warning: /opt/rh/devtoolset-7/root/usr/bin/g++: linker input file unused because linking not done
/opt/rh/devtoolset-7/root/usr/bin/gcc /opt/rh/devtoolset-7/root/usr/bin/g++ -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include     -std=gnu++14 -I"/usr/share/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/R/x86_64-pc-linux-gnu-library/4.0/BH/include'    -fpic  -g -O2 -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -g  -c -c LASlib/laswriter_las.cpp -o LASlib/laswriter_las.o
gcc: warning: /opt/rh/devtoolset-7/root/usr/bin/g++: linker input file unused because linking not done
/opt/rh/devtoolset-7/root/usr/bin/gcc /opt/rh/devtoolset-7/root/usr/bin/g++ -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include     -std=gnu++14 -I"/usr/share/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/R/x86_64-pc-linux-gnu-library/4.0/BH/include'    -fpic  -g -O2 -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -g  -c -c LASlib/lasreaderbuffered.cpp -o LASlib/lasreaderbuffered.o
gcc: warning: /opt/rh/devtoolset-7/root/usr/bin/g++: linker input file unused because linking not done
/opt/rh/devtoolset-7/root/usr/bin/gcc /opt/rh/devtoolset-7/root/usr/bin/g++ -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -I/usr/local/include     -std=gnu++14 -I"/usr/share/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/R/x86_64-pc-linux-gnu-library/4.0/BH/include'    -fpic  -g -O2 -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -g  -c -c LASlib/lasreadermerged.cpp -o LASlib/lasreadermerged.o
gcc: warning: /opt/rh/devtoolset-7/root/usr/bin/g++: linker input file unused because linking not done
/opt/rh/devtoolset-7/root/usr/bin/gcc /opt/rh/devtoolset-7/root/usr/bin/g++ -fPIC  -I"/opt/R/4.0.2/lib/R/include" -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/inc [... cut ...]
15: try(suppressPackageStartupMessages(library(pkg_name, lib.loc = lib,     character.only = TRUE, logical.return = TRUE)))
16: tools:::.test_load_package("rlas", "/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/00LOCK-rlas/00new")
An irrecoverable exception occurred. R is aborting now ...
ERROR: loading failed
* removing ‘/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/rlas’
Warning in install.packages :
  installation of package ‘rlas’ had non-zero exit status

The downloaded source packages are in
	‘/tmp/RtmpIPjKaS/downloaded_packages’
```

Jean-Romain, 2021-04-27:
you have twice `-c` remove `-c` from `Makevars`

mxblsdl, 2021-04-27:
I removed duplicates from the Makevars which now looks like this:
```
CXX14 = /opt/rh/devtoolset-7/root/usr/bin/gcc -fPIC
CXX14FLAGS = /opt/rh/devtoolset-7/root/usr/bin/g++ -std=gnu++14 -DNDEBUG -DNDEBUG -DUNORDERED -DHAVE_UNORDERED_MAP -I./ -I./LASlib/ -I./LASzip/ -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/Rcpp/include' -I'/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/BH/include' -g -O2 -fdebug-prefix-map=/build/r-base-tRgc13/r-base-4.0.2=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -c
```

I can compile the package with the warning:
`gcc: warning: /opt/rh/devtoolset-7/root/usr/bin/g++: linker input file unused because linking not done`

But then the installation errors with: 
```
** building package indices
** testing if installed package can be loaded
Error: package or namespace load failed for ‘rlas’ in dyn.load(file, DLLpath = DLLpath, ...):
 unable to load shared object '/home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/rlas/libs/rlas.so':
  /home/mblasdel/R/x86_64-pc-linux-gnu-library/4.0/rlas/libs/rlas.so: undefined symbol: _ZSt24__throw_out_of_range_fmtPKcz
Error: loading failed
Execution halted
```

I think this has to do with the `g++` library. The `g++` was installed in the same manner as `gcc` with a devtools set for RHEL. I had to run `scl enable devtoolset-7 bash` which is why I use the specific file path in the `CXXFLAGS`.

Jean-Romain, 2021-04-27:
At this stage I can no longer help you. I tried to reproduced CRAN like compilation with you but at this stage there are too much Red Hat specific stuff. Why R does not compile with all the default option? Why gcc is installed in a special way? Why code definitions are missing (this is the current error)? Too many question and I can't reproduce.

mxblsdl, 2021-04-27:
Fair enough. I really appreciate your help on this. 

`gcc` needs to be installed in a special way because the default `gcc` that ships with RHEL is an older version. Hence the need for the `devtoolset-7` install. I'll keep trying to work through it with my companies IT. 

Again, `lidR` is a great package and I've been able to process data on a local laptop, but the RHEL env is proving difficult to work in.

Jean-Romain, 2021-04-28:
For your information the next version of R (4.1.0) that is coming in April will start compiling in C++14 by default. See https://cran.r-project.org/doc/manuals/r-devel/NEWS.html

> The default C++ standard has been changed to C++14 where available (which it is on all currently checked platforms): if not (as before) C++11 is used if available otherwise C++ is not supported. 

So, Red Hat user like you will have to del with that.

closed 2021-05-06 by Jean-Romain