r-lidar/rlas #53: Unable to load las or laz files

denzle83, 2021-07-04:
Hi, 

I'm having trouble opening my las files. Where just last week everything was fine, now nothing seems to load. I've moved files from my own dir to the lidR one, have compressed the files to LAZ and tried every conceivable method to load them but either receive:

- Error: File does not exist.
or 
- Error: Invalid header: 'description' must be a string of length < 32 in extra bytes description 

I've searched for any information relating to the 'Invalid header' but find nothing of help. I have updated R and RStudio to the latest as well uninstalling and re-installing everything and installing an older version. Also renaming the files to see if that has any impact but nothing seems to work. Do you have any ideas what could be the cause?

Many thanks
Dan

Jean-Romain, 2021-07-04:
Two workaround in `lidR` before I find a fix.

## Do not read extra bytes

This way the header will be modified to remove information about extra bytes and it will consequently not trigger any error

```r
las = readLAS("hen_faes_plot3_lidar.laz", select = "-0")
```

## Create a `LAS` object manually

```r
p = rlas::read.las("~/Téléchargements/hen_faes_plot3_lidar.laz", filter = "-keep_random_fraction 0.01")
h = rlas::read.lasheader("~/Téléchargements/hen_faes_plot3_lidar.laz")
h$`Variable Length Records`$Extra_Bytes$`Extra Bytes Description`$NormalZ$description = "Z component of a vector normal t"
las = LAS(p, h)
```

denzle83, 2021-07-04:
Thank you for the work arounds they are a great help and greatly appreciated.

Dan

Jean-Romain, 2021-07-05:
The bug is expected to be fixed. Please install rlas 1.5.1 and tell me if it works well

commit 7f0ca269 (2022-02-09): Fix #53  global-buffer-overflow on CRAN
closed 2021-10-20 by Jean-Romain