r-lidar/rlas #54: Error: Invalid data: ScanAngleRank is not an integer on 8 bits

Robin-hartley, 2021-08-04:
Hi there JRR,

I've been working on a bit of Riegl MiniVUX data for the past 6 months, had a nice script running in LidR to derive some vegetation metrics etc, and then was unhappy with the noise in the data, so I created a pipeline in LASTools to denoise the data prior to entry into Lidr. Unfortunately, I am now getting this error message when I try to ground classify the data:

```r
plan(multisession)
opt_output_files(ctg) <- paste0("D:/FILEDIRECTORY/{*}_gc")
ctg_gc <- classify_ground(ctg, csf())
#> Processing [>------------------------------------------------------]   1% (1/81) eta:  4m
#> Error: Invalid data: ScanAngleRank is not an integer on 8 bits
```

I assume this is something to do with the format of the ScanAngleRank data that is written to the LAS file by LASTools. The same happens when I export data from CloudCompare. 

I have a small snippet of the data I am working with here -
[RH_test_data.zip](https://github.com/Jean-Romain/lidR/files/6927914/RH_test_data.zip)

Are you able to reproduce this error, and is it a bug? Or do I need to somehow edit the LAS file? I saw a bit of code that looked potentially useful here: https://github.com/Jean-Romain/lidR/issues/319 in which you suggested changing the ScanAngleRank data to an integer:

```r
las@data[["ScanAngleRank"]] <- as.integer(las@data[["ScanAngle"]]) # If prf >= 6
```

However, when I tried this on the test data I have provided, well, I couldn't load it into lidR as the same error appeared.

Thanks in advance,

RH

Jean-Romain, 2021-08-04:
The issue is from `rlas`. I transfered the issue. The value -128 triggered a false positive error. Update `rlas 1.5.1` from github.`You now have only warnings

```r
 las = readLAS("~/Téléchargements/issue 454/RH_test_data.laz")
#> Warnings :
#> 1: Invalid data: ScanAngleRank greater than -90 degrees. 
#> 2: Invalid data: ScanAngleRank greater than 90 degrees 
#> 3: Invalid data: 66 points with a 'return number' greater than the 'number of returns'. 
```
```

closed 2021-08-04 by Jean-Romain

Robin-hartley, 2021-08-06:
Many thanks for getting to this so promptly

commit 35994063 (2021-08-04): Fix #54