r-lidar/rlas #56: R crashes when reading in laz files with lidR

AndresKasekamp, 2022-04-27:
I have problem with reading in laz files from one of the flight years from my NMA. It causes RStudio to crash without any messages. All the other years work perfectly and I can easily apply functions to it. Using the erroneous laz files in LASTools has not produced any errors. I am using the latest lidR version.

It is probably related to https://stackoverflow.com/questions/60457539/lidr-readlas-function-crashes-r-when-reading-some-las-files . 

I am attaching a Google Drive link with the error and no error laz files.
https://drive.google.com/drive/folders/1SVsfenH_petxogLg7-QvBaeQ98aHrDvj?usp=sharing

Jean-Romain, 2022-04-27:
This is not related to `lidR`. Moved to `rlas`

```r
library(rlas)
las = read.las("issue 574/no_error.laz")
```

AndresKasekamp, 2022-05-03:
I managed to solve the problem. The erroneous laz had Extra Byte attributes "Amplitude" and "Pulse width". The no error laz had "Amplitude", "Pulse width", "Reflectance" and "Deviation". They are numbered 1-4 accordingly. By specifying which attributes to read in (or not to read in), RStudio did not crash anymore.
For example, las <- readLAS(r"(laz_check\error.laz)", select = "xyzia")

LidR reads every attribute in by default. However, it it is interesting that my error.laz had attributes "Amplitude" and "Pulse width", but I cannot use them. The error.laz "Amplitude" did have a different type than no_error.laz (4 (short) vs 3 (unsigned short)). However, "Pulse width" had the same data type and using "xyzia2" still causes RStudio to crash.