r-lidar/rlas #59: lidR::readLAS for most las files even lesser 50KB is crashing R

nidhi23ug, 2023-01-22:


Jean-Romain, 2023-01-22:
`rlas::read.las("~/Téléchargements/08nd0862.dams__clipped_polygon_2.las")`

Confirmed

Jean-Romain, 2023-01-22:
Seems to be related to extrabytes

```r
# works
rlas::read.las("~/Téléchargements/08nd0862.dams__clipped_polygon_2.las", select = "")
rlas::read.las("~/Téléchargements/08nd0862.dams__clipped_polygon_2.las", select = "* -0")

# crashes
rlas::read.las("~/Téléchargements/08nd0862.dams__clipped_polygon_2.las", select = "1")
```

nidhi23ug, 2023-01-22:
great this works. Thank you

Jean-Romain, 2023-12-07:
I don't where I put the file. Did you send it to me by email. What is your name so I can find your email.

dimfalk, 2024-06-13:
> [nidhi23ug](https://github.com/nidhi23ug) commented on Jan 21, 2023 via email

Yup, the mail should have the file `08nd0862.dams__clipped_polygon_2.las` attached.

Jean-Romain, 2024-06-13:
That makes sense... I found it!