r-lidar/rlas #68: Fail to clip from LAScatalog with both copc and non-copc laz files

jmmonnet, 2024-03-08:
I have a LAScatalog with files which are either copc or not copc. I noticed that when using parallelisation, some threads ran forever, so I tried to figure out in which cases and when the process does not end. I ended up with one example. `clip_circle` succeeds with a LAScatalog with two files (one copc, on not copc) depending on the order in which they appear in the catalog. 

- test.laz is available from [https://filesender.renater.fr/?s=download&token=7368c29e-8e70-4c8f-9af4-1e2090f70f45](https://filesender.renater.fr/?s=download&token=7368c29e-8e70-4c8f-9af4-1e2090f70f45)
- LHD_FXX_0726_6357_PTS_O_LAMB93_IGN69.copc.laz from [https://storage.sbg.cloud.ovh.net/v1/AUTH_63234f509d6048bca3c9fd7928720ca1/ppk-lidar/LO/LHD_FXX_0726_6357_PTS_O_LAMB93_IGN69.copc.laz](https://storage.sbg.cloud.ovh.net/v1/AUTH_63234f509d6048bca3c9fd7928720ca1/ppk-lidar/LO/LHD_FXX_0726_6357_PTS_O_LAMB93_IGN69.copc.laz). I am sorry I could not produce a smaller copc file.

Those files do not have the same CRS information in the header although they share the same projection

Catalog files with non-copc first
`cata <- lidR::catalog(c("test.laz", "LHD_FXX_0726_6357_PTS_O_LAMB93_IGN69.copc.laz"))`
clip_circle runs ok
`pc <- lidR::clip_circle(cata, 726000, 6357000, 20)`

Catalog files with copc first
`cata <- lidR::catalog(c( "LHD_FXX_0726_6357_PTS_O_LAMB93_IGN69.copc.laz", "test.laz"))`
clip_circle does not end on my machine (ubuntu)
`pc <- lidR::clip_circle(cata, 726000, 6357000, 20)`

I have lidR 4.0.4 and rlas 1.7.0 from CRAN. I am a bit confused because in some other examples I tested, unexpected behaviours occured (e. g. with 2 non-copc files plus 2 copc files in a LAScatalog, `clip_circle` does not succeed when lax files are present for the non-copc files, but `clip_circle` succeeds when the lax files are removed...)

Jean-Romain, 2024-03-08:
Minimal reproducible example

```r
setwd("/home/jr/Téléchargements/lidR issues/issue 744")
f = c("test.laz", "LHD_FXX_0726_6357_PTS_O_LAMB93_IGN69.copc.laz")
rlas::read.las(f, filter = "-inside_circle 726000 6357000 20")
rlas::read.las(rev(f), filter = "-inside_circle 726000 6357000 20")
```
Moving the issue to `rlas`

Jean-Romain, 2024-03-12:
Note to myself: case 3 does not work either it reads only `test.laz` and return 2255 points

Jean-Romain, 2024-03-12:
Fixed. It was a bug in LASlib itself

closed 2024-03-12 by Jean-Romain

jmmonnet, 2024-03-14:
Thank you very much for investigating this issue. If I want lidR to correctly handle LAScatalog with both types of files, should I build `rlas` with a fixed version of `LASlib` ? what would be the procedure on a linux machine ?

Jean-Romain, 2024-04-15:
By fixing this bug I created a new one. I fixed it. Please try again to ensure that the initial bug is still fixed.

jmmonnet, 2024-04-16:
I tested with two other files and was not able to reproduce the bug (I am sorry I could not use the same files as in the previous example because I deleted the "test_laz" one).

commit cc6f5113 (2024-03-12): Fix #68
commit c0988e70 (2024-04-15): Fix #68