r-lidar/rlas #69: `readLAS()` leads to RStudio crash when applied on a corrupt file

dimfalk, 2024-03-27:
When applying `readLAS()` on a corrupted file, {lidR} causes the IDE to crash at the moment:

``` r
library(lidR)
#> 4.1.1

f <- "3dm_32_299_5634_1_nw.laz"

utils::download.file(paste0("https://www.opengeodata.nrw.de/produkte/geobasis/hm/3dm_l_las/3dm_l_las/", f), 
                     f)

lidR::readLAS(f)
#> --- CRASH ---
```
[image]

Meanwhile, I'm fully aware that this is due to the wrong `mode` in `utils::download.file()`, but I feel like a package should not cause a crash of the superordinate piece of software and also that it would have been nice to get a meaningful error message, in order to be able to investigate further instead of the upper notification prompted by RStudio itself.

``` r
f <- "3dm_32_299_5634_1_nw.laz"

utils::download.file(paste0("https://www.opengeodata.nrw.de/produkte/geobasis/hm/3dm_l_las/3dm_l_las/", f), 
                     f,
                     mode = "wb")

lidR::readLAS(f)
#> class        : LAS (v1.2 format 1)
#> memory       : 517.8 Mb 
#> extent       : 299000, 3e+05, 5634000, 5635000 (xmin, xmax, ymin, ymax)
#> coord. ref.  : ETRS89 / UTM zone 32N 
#> area         : 1 km²
#> points       : 8.48 million points
#> density      : 8.48 points/m²
#> density      : 8 pulses/m²
```

Just to show how other packages handle this: 
When messing things up using e.g. {terra}, i.e. GDAL, you get the following error message. Something similiar would be nice, I guess.

``` r
f <- "snw_NAM-44_CCCma-CanESM2_historical-r1_r1i1p1_CCCma-CanRCM4_r2_mon_195001-195012.nc"

download.file(url = paste0("https://crd-data-donnees-rdc.ec.gc.ca/CCCMA/products/CanSISE/output/CCCma/CanRCM4/NAM-44_CCCma-CanESM2_historical-r1/mon/atmos/snw/r1i1p1/", f),
              destfile = f)

terra::rast(f)
#> Warning:
#> `snw_NAM-44_CCCma-CanESM2_historical-r1_r1i1p1_CCCma-CanRCM4_r2_mon_195001-195012.nc'
#> not recognized as a supported file format. (GDAL error 4)
#> Error: [rast] cannot open this file as a SpatRaster: snw_NAM-44_CCCma-CanESM2_historical-r1_r1i1p1_CCCma-CanRCM4_r2_mon_195001-195012.nc
```

Jean-Romain, 2024-03-27:
Transferred to `rlas`, reproduced under windows

```r
f <- "3dm_32_299_5634_1_nw.laz"
utils::download.file(paste0("https://www.opengeodata.nrw.de/produkte/geobasis/hm/3dm_l_las/3dm_l_las/", f),  f)
rlas::read.las(f)
```

dimfalk, 2024-03-27:
Just wanted to confirm that RStudio still crashes using R 4.3.3 and `rlas::read.las()`.

ColinPSweeney, 2024-08-08:
Just encountered a similar issue using Mac (14.3.1), R (4.4.0), rlas (1.7.0), lidR (4.1.1). Seems to be not just a Windows issue. As with the case for @dimfalk , it is just specific corrupt files. The rest of the ones I am using work normally for me and don't crash R.